lincomatic/open_evse #23: Sleep status cause problem on Smart ED

martingraml, 2015-02-19:
What steps will reproduce the problem?
1. Connect Smart ED to EVSE
2. Set EVSE on via RAPI "FE"
3. Set EVSE sleeping "FS"
4. Set EVSE on via RAPI "FE" within 60 sek

Smart ED LED on Charging Socket flasching red.
I think the problem is that the EVSE goes to status disabled and disconnects the relais.

What is the expected output? What do you see instead?

when EVSE is set to sleeping, don´t open the relais and set state to disabled. Sleeping means, that the it is possible to start the charging immediately.
Maybe it is useful to make a new Command on Rapi -> "Soft Stop"

What version of the product are you using? On what operating system?
OpenEVSE Plus DIY

lincomatic, 2015-02-20:
The whole purpose of the "sleep" state is to let you have the car connected w/o charging. What would be the use of sleep state w/o opening the relay?

lincomatic, 2015-02-25:
Since you haven't replied in 5 days, I'm going to assume that this issue needs no further attention.

closed 2015-02-25 by lincomatic

martingraml, 2015-02-25:
Sorry for the late answer.

I have a photovoltaik and a smart meter at home.
So I can control the EVSE via Remote according the power available (measured with the smart meter)

When the consumtion exeeds the production of the photovoltaic, i set the EVSE to state sleep.
Now the relais opens. When i start loading again the smart ED goes into an error status.
I think i can this avoid when the relais doesn�t open - only when the car is fully chared or i want a stop status the relais should open.

If i charge the car only with the current set until the car is fully charged this is no problem.

For my purpose i also changed that the current capacity is saved in the eeprom on every change.

Best regards

lincomatic, 2015-02-26:
I still don't understand how you want the Sleep state to function. There is no such state specified in J1772. So I created this state. Relays are supposed to be closed ONLY in STATE C (charging state) according the the spec. So my "sleep state" is to turn off PWM (pilot is +12V) and open the relay. Any other method would violate J1772. If you wish to modify your personal firmware to do otherwise, you are free to do so, but I can't change it in the public version.