lincomatic/open_evse #52: RAPI GS command response and ST asynchronous have different number format

presslab-us, 2016-01-03:
For example, in sleep mode the RAPI GS command response would be as follows: "OK 254 1".  The asynchronous state announce would be like this: "ST fe".  The documentation in rapi_proc.h implies they are in the same format "EVSE_STATE_xxx".

Perhaps the decimal response is better than hexadecimal because that's what everything else uses.  But I do think they should have the same number format.  What do you all think?

lincomatic, 2016-01-16:
Good point. While I agree that it's an inconstency (sloppiness on my part), I'm not sure it's worth breaking existing code to fix it.

presslab-us, 2016-01-30:
Yes, probably best to leave it.  If I get some free time I'll try to clarify the documentation around this.

lincomatic, 2019-06-27:
This is solved by $AT notification, which replaces $ST, and uses hex format.. code currently only resides in ocpp branch

closed 2019-06-27 by lincomatic