lincomatic/open_evse #53: RAPI response has no checksum

presslab-us, 2016-01-06:
There doesn't seem to be any checksum ability on the RAPI response packet.  Maybe this was deemed unnecessary as the responses won't cause potential hazards like a bad command could.  I'm just wondering what the thoughts on this are; if I added a way to optionally turn on a response checksum (XOR) would that be desired?

lincomatic, 2016-01-16:
This is a good idea. It was an oversight on my part not to add a checksum to the response. Unfortunately, to avoid breaking existing code, as you mentioned, it should be optional, so I guess we need a command to enable/disable it.

lincomatic, 2016-01-26:
The latest dev release (D4.0.0) now appends an XOR checkbyte to all RAPI responses. At the risk of breaking existing code, I decided to just enable it by default. It will take too much code space to add a command to disable it, so it can only be disabled by commenting out #define RAPI_RESPONSE_CHK.

Thanks again for bringing this glaring omission to my attention.

closed 2016-01-26 by lincomatic

presslab-us, 2016-01-28:
Great!  I will test it with my ZigBee module and let you know how it goes.

lincomatic, 2016-02-02:
I noticed another omission today.. the async messages $ST and $WF were also missing check bytes. I just pushed vD4.1.0, which adds them. .. new RAPI version is 2.0.1

presslab-us, 2016-02-20:
I updated my ZigBee code to work with the checksum, and everything worked as expected.  Thanks.