lincomatic/open_evse #54: Temperature monitoring - subzero temps

MortenSo, 2016-01-08:
Apologies if this is the wrong place to note this. In freezing temperatures, the temperature control triggers an overtemperature alarm and stops charge. Does the temp sensor work sub zero?

lincomatic, 2016-01-16:
This is a good place to bring this up. I will defer to craigk to look into this when he gets back from his travels. In the meantime, you can disable temperature monitoring by commenting out
#define TEMPERATURE_MONITORING
in open_evse.h

MortenSo, 2016-01-17:
I disabled through the settings menu :-) I think I saw a temperature reading of 255 (which I guess indicate overflow or outside of range reading) before it went to over temperature fault. (May have been in my own version of the software I saw this reading - don't quite remember - the behaviour is nevertheless the same in the latest stable release).

lincomatic, 2016-01-17:
Oops, I forgot that Craig added it to the menu

lincomatic, 2016-01-26:
The bug was caused by improper handling of negative temperatures. It has been fixed. Please download v3.11.3.

https://github.com/lincomatic/open_evse/releases/tag/v3.11.3

closed 2016-01-26 by lincomatic