lincomatic/open_evse #55: Compile Error

andyfras, 2016-01-14:
Newbie here; sorry if this has already been asked. When I try to compile the latest download, I get the following error:

> In file included from sketch\ammeter_cal\ammeter_cal.cpp:23:0:

> sketch\ammeter_cal\ammeter_cal.h:29:18: fatal error: time.h: No such file or directory

>  #include <time.h> (time.h)

Please note that time.h won't display with the correct brackets on here, so I've substituted round ones.

lincomatic, 2016-01-14:
which compiler are you using? This is a standalone C++ program, not Arduino sketch

andyfras, 2016-01-15:
Yes, I was using the Arduino compiler. I've obviously still got a lot to learn.

I want to build an EVSE inside a type 2 Mennekes connector using an Arduino Pro Mini.

Thanks for your patience.

closed 2016-01-16 by lincomatic