lincomatic/open_evse #66: Setting the LCD text via RAPI can corrupt display

jeremypoulter, 2017-07-21:
When setting the LCD text via RAPI (eg '$FP 0 0 Client-IP.......') the display can get corrupted depending on the state of OpenEVSE, eg;

[image]
[image]

Could we add an RAPI command to claim/release the the LCD? When claimed the LCD will only be updated via RAPI (using $FP, $FB). On release the LCD will revert back to OpenEVSE control.

glynhudson, 2017-07-24:
I think the same issue has been reported here, it seems the openevse FW stable production branch is on the openevse github: https://github.com/OpenEVSE/open_evse/pull/4

lincomatic, 2017-07-25:
Please have a look at the $F0 command in the dev branch. It does what you need

closed 2017-07-26 by lincomatic