lincomatic/open_evse #67: Async event for setting change

jeremypoulter, 2017-07-21:
It would be really handy to send an async event when changes are made to any of the settings via the LCD, eg ```$VC GE``` if any of the flags change. This would cut down on a lot of the polling the WiFi module has to do.

lincomatic, 2017-07-27:
we're running out of code space, no room to do anything fancy. maybe just send an async message whenever exiting menu mode?

jeremypoulter, 2017-07-27:
That could work, thinking about it maybe an event on entering the menu would also be handy so OpenEVSE WiFi can block making changes remotely while someone is using the local UI.

lincomatic, 2019-06-27:
OK, via the $AT notification, which replaces the $ST notification.
'''
$AT evsestate pilotstate currentcapacity vflags
 evsestate(hex): EVSE_STATE_xxx
 pilotstate(hex): EVSE_STATE_xxx
 currentcapacity(decimal): amps
 vflags(hex): m_wVFlags bits
'''
There is a new vflag that tells you when the UI is in a menu:

```
#define ECVF_UI_IN_MENU         0x0800 // onboard UI currently in a menu
```
Note that this code is currently only in the ocpp branch.

closed 2019-06-27 by lincomatic

jeremypoulter, 2019-06-27:
Thanks, I will take a look.