lincomatic/open_evse #73: L2 minimum current should be 6A not 10A?

glynhudson, 2017-10-19:
The correct L2 minimum current should be 6A not 10A:

https://github.com/lincomatic/open_evse/blob/development/firmware/open_evse/open_evse.h#L354

Updating from OpenEVSE v3.10.4 to v4.8.0 the minimum L2 current increased to 10A. I've traced the change back to this commit: https://github.com/lincomatic/open_evse/commit/e144329e2965ca85369f9825ea6332839ab9e0bf

I can't find any comments about why the minimum current was increased to 10A? Trickle charging at 6A used to work fine

*** 

On a slight different topic: IMO it would also be a good idea to limit L2 max current to 40A max since this is the maximum the single-phase cables EV cables can handle. Most are rated at 32A. It would certainty be a good idea to limit the max rate to 40A for a full-assembled 40A unit. This issue should probably be opened with @chris1howell  on the OpenEVSE repo.

glynhudson, 2017-10-31:
Can you confirm this is a bug rather than an intentional change? Is it possible to release an updated version that allows 6A to be selected?

chris1howell, 2017-10-31:
It is just a setting in openevse.h. I will change MIN_CURRENT_CAPACITY_L2 in 4.8.0 back to 6A as defined by SAE J1772.

// minimum allowable current in amps
#define MIN_CURRENT_CAPACITY_L1 6
#define MIN_CURRENT_CAPACITY_L2 6

// maximum allowable current in amps
#define MAX_CURRENT_CAPACITY_L1 24 // J1772 Max for L1 on a 20A circuit
#define MAX_CURRENT_CAPACITY_L2 40 // J1772 Max for L2

glynhudson, 2017-10-31:
>It is just a setting in openevse.h. I will change MIN_CURRENT_CAPACITY_L2 in 4.8.0 back to 6A as defined by SAE J1772.

Fantastic :+1: . It would be a good idea to also bump the version to `4.8.1` to distinguish between the two. 

Are you also going to set 40A maximum on your 40A units? I would be happy to receive units with 40A max setting this since all the units we ship are 32A max (UK single-phase recommended limit).

glynhudson, 2017-11-03:
I see this issue has been fixed by https://github.com/lincomatic/open_evse/commit/48235616a3ed5be6fe438bca633565adc30b67eb thanks @lincomatic 

@lincomatic are there any known issues with using the development branch? I'll try and compile and test. Thanks :+1:

closed 2017-11-14 by glynhudson