lincomatic/open_evse #75: LCD RGB backlight colour changes

glynhudson, 2017-11-11:
Recently the background LCD colour when the EVSE is sleeping was changed: https://github.com/lincomatic/open_evse/commit/984857f33d372ddd21cbf3f59ccbf4033f2314c6

Now the colour when sleeping is identical to the current when charging. Is this intentional? It was quite useful to be able to distinguish between sleeping / charging at a glance. 

[image]
[image]

lincomatic, 2017-11-11:
Actually, the colors are still different. 
charging: TEAL
sleeping (disconnected): VIOLET
sleeping(connected): WHITE

The rationale for the change: If you have the timer turned on, and the EVSE is sleeping when you plug in the EV, there's no indication that you plugged the cable into the car correctly.
It's easy to see when the EVSE is awake, because it goes from waiting to charging modes, and changes color. 
Before the change, the EVSE would be VIOLET whether or not it was connected when sleeping.
Now, it's VIOLET when sleeping and when you plug in the EV, it turns white, giving you a positive visual sign that the EV is plugged in correctly, so that you don't get an unpleasant surprise, and find a dead battery in the morning.

Your display has bad color balance, and is not properly displaying WHITE. if you can think of a better color to use that doesn't clash with existing colors, let me know, and I will try it.. the code is in OnboardDisplay::Update()

glynhudson, 2017-11-13:
Ok, that makes sense. The issue I have as you can see from the photo above on my OpenEVSE WHITE and TEAL look exactly the same. I have tested this on a number of different units all using standard OpenEVSE RGB LCD's obtained directly from OpenEVSE store. 

I had not realised that the display is VIOLET when sleeping and disconnected, that's good. 

It's not a big issue. I can live with it. Are the full colour codes documented anywhere? 

What about BLUE while charging? Is blue used anywhere?

lincomatic, 2017-11-14:
Actually, in your photos above, I can easily tell the difference between WHITE and TEAL, even though there's a strong blue cast to the WHITE.

The colors are all in open_evse.h

// for RGBLCD
#define RED 0x1
#define YELLOW 0x3
#define GREEN 0x2
#define BLUE 0x4
#define TEAL 0x6
#define VIOLET 0x5
#define WHITE 0x7 

I can't remember why I didn't use blue, because it was a long time ago, but probably because it looked dark and low contrast on my display. If you find a color combo that works, feel free to propose it here, and I'll see if it looks good on my displays.

The code is pretty straightforward in OnboardDisplay::Update().

lincomatic, 2017-11-21:
Do you have any better color combination to test, or should I just close this?

glynhudson, 2017-11-22:
No, I've not been able to fine a bitter combination. Best close it for now.
It's not a big issue, I've got used it it now :-)

closed 2017-11-22 by glynhudson