lincomatic/open_evse #76: Session Wh counter does not reset if using delay timer

glynhudson, 2017-11-14:
The Wh session counter does not reset in between charging sessions when the delay timer is active.

Example: I use the OpenEVSE charging timer to schedule overnight charging of an EV, after a few days the session counter is massive and therefore useless as a "session counter". This also makes the "Energy Limit" feature impossible to use. 

See session wh counter screenshot below showing of 35 kWh which is impossible for a single session since my 24kWh Nissan LEAF only has 20 Kwh of usable capacity! Also I'm pretty sure my domestic single phase-supply would struggle to deliver 25kWh over elapsed time of 10min! 

[image]

We it be possible to reset the session counter (write it to total kWh EEPROM) when state changes from:

`Charging `

to 

`SLEEPING (connected) or  SLEEPING (disconnected) or READY`

lincomatic, 2017-11-17:
Hmm.. The current behavior was implemented because it was assumed that the SLEEP function would be engaged either by pressing the switch or via RAPI.. the assumption was that SLEEP was just a pause in the current charging session. 

It seems more logical to me to terminate a charging session when either the EVSE goes back to READY, or if the EV is disconnected. What do you think?

BTW, the Wh counter isn't even accurate as currently implemented:
https://github.com/lincomatic/open_evse/issues/77

lincomatic, 2017-11-21:
this should fix it.
https://github.com/lincomatic/open_evse/commit/9ee6f1c1579dda2e98664debbbe8e34d2b678b68
https://github.com/lincomatic/open_evse/commit/ffc8e535e401a23b7d8e31c7be2534f681757663

The total now gets updated/written to EEPROM when the EV disconnects, and session counter resets when the EV connects again

Please test. If it's still not behaving properly, you can reopen this

closed 2017-11-21 by lincomatic

glynhudson, 2017-11-24:
Fixed! Works great, thanks @lincomatic 👍