lincomatic/open_evse #79: Session elapsed time should be consistent with session energy

glynhudson, 2017-12-01:
I'm running latest dev version with the update to reset the session energy when the car in un-plugged even in the unit is in 'sleep' state (charging timer active). This is working very well.

However I have noticed another issue: the time elapsed resets each time the evse is paused then resumed. IMO the session timer should be consistent with session energy. I would consider pausing then resuming a charge (without un-plugging the car) the be the same charging session therefore the time elapsed should not reset. See illustration video: 

https://youtu.be/R8hdGba5Ygw

lincomatic, 2017-12-04:
I agree.

https://youtu.be/R8hdGba5Ygw

lincomatic, 2017-12-08:
OK, it's fixed. We now reset the session counter only when the EV is reconnected after it's been disconnected, so it's consistent w/ the energy counter

https://github.com/lincomatic/open_evse/commit/ee0f1b3f49d6884936c4c267d97b2f340dbfea03

Thanks for bringing this to my attention.

closed 2017-12-08 by lincomatic

glynhudson, 2017-12-14:
Fantastic, thanks :+1: