lincomatic/open_evse #80: Set temporary (single session only) charging current

glynhudson, 2017-12-28:
As far as I'm aware using the volatile flag when setting charging current e.g `$ SC [amps] V` only restores the original current after a reboot. 

Would it be possible to add a RAPI flag to set a temporary (single session only) current? After the session ends (car is unplugged) the current would restore back to default.

Maybe a S flag could be used for this to denote session? e.g. 

`$ SC [amps] S`

lincomatic, 2017-12-29:
Could you help me understand the use case for this feature ?

glynhudson, 2018-01-01:
> PgaJpZM4ROd8v>.

It's a common use case of users on site renewable generation or variable
available power to want to charge at a particular rate for a particular
season e.g user is home when solar pv is generating therefore set charging
rate to 6A/10A for that charging session. Then when another session is
started later in the evening the default (faster) charging rate should he
restored to allow the EV to charge  normally again.

Or the opposite: a user may choose to charge at a higher rate than usual
since they are desperate for a quick topup charge, this charging current
may not be suitable for normal use if other applications are running.

The final usecase I've come across is users using the openevse as a
portable evse therefore the charging current is set at each location
depending on what electrical outlet the EVSE is connected to.

- sent from my mobile device

lincomatic, 2018-02-08:
Couldn't this be implemented on the RAPI client side? A notification is sent out whenever the EVSE changes state. The RAPI client could act upon this and reset the max current using the existing framework.

glynhudson, 2018-02-08:
Yeah sure, it could be. There are more important features more worthy implementing e.g #81 
I will close this issue.

closed 2018-02-08 by glynhudson