lincomatic/open_evse #81: Alert user when temperature throttling was activated during a charge session

glynhudson, 2018-02-06:
Currently the RGB LCD displays red when temperature throttling is active i.e charging rate is reduced if temperature goes above 75 degC:

```
50% | 65°C - 150°F
25% | 68°C - 155°F
Shutdown | 71°C - 160°F
Resume 100% | 62°C - 145°F
```

Temperature throttling is great and is an important safety feature however it can have the side effect of masking an underlying issue with the EVSE build that's causing excessive heating e.g. EVSE starts to overheat (up to 70 deC) > charge current is throttled > temperature plateaus > temp never reaches the 71 deg C shutdown level > but damage to components still occurs overtime. 

Especially in cold (sub-zero) ambient temperatures the actual conductors get very hot indeed before triggering a over temp shutdown. In many parts of the world in winter 70 deg C is actually rather hot and defiantly not normal. 

Most users charge their EV overnight in a garage therefore would miss spotting the instant the LCD turn red to indicate high temp and charge rate throttled. The user would wake-up and unplug their car in the mornig with the EVSE showing a normal display unaware of issue causing the overheating. 

A possible solution to this suggested by @chris1howell would be to display a warning message on the second line of the LCD which is persistent until the car is unplugged to alert the user that at some point in the current charging session temperature throttling was activated. The user can then investigate accordingly. The LCD could also display red until car is unplugged. 

Do you think this is a viable solution that should act as an early warning for potential build issues? I understand that healthy systems in hot countries could potentially trigger this warning under 'normal' operation, however the warning could still be useful since the user is alerted that the car might not be fully changed since charging rate was throttled during the session. Happy to discuss other options...

lincomatic, 2018-02-06:
Sounds like a great idea. Unfortunately, I've been fixing subtle issues w/ the delay timer for several days, and won't have time to work on this for a while. We should probably send some sort of notification or at least save some sort of temperature trip count in EEPROM, as well

lincomatic, 2018-02-08:
I'm trying to figure out a quick & dirty way to do something that will be useful. I recall that the screen displays temperature and blinks red or something like that during overtemp? Do you have some photos
of what's displayed?

I guess I could make the display stay red and display something until the EV gets disconnected.

The question is, how to make the information truly useful. It seems like knowing how long the event happened and possibly the highest temperature? Or what if it happened multiple times during a session.
We should come up with a way for the user to understand the severity.

glynhudson, 2018-02-08:
> I guess I could make the display stay red and display something until the EV gets disconnected.

Yes, I think that's all that's needed. This would alert the user to investigate the issue further e.g look at the Emoncms datalog to see exactly how hot and how long. It would get very complicated vary quickly trying to display this on the LCD! 

e.g. here is a trace showing the temperature reaching almost 70 degrees (right axis) and the charge rate being throttled 

[image]

lincomatic, 2018-02-08:
Ah, right, I forgot about emoncms. I guess that's all we need to investigate further.
I've been having trouble with running your ESP code. The old version would just freeze up after a few hours. I tried to compile the latest last night, and my ESP just goes into outer space

glynhudson, 2018-02-08:
What old version where you running. Hanging issues were fixed in V2.4.0 in September 2017. The latest stable release is V2.5.2. This is very stable and run well on ESP12: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases 

Make sure you fully erase the esp using esp tool before uploading the new version. The new version has the file system built into the flash so only requires a single upload and it easy to upload OTA via HTTP web uploader. 

If you want to checkout the interface I have a demo up and running here: http://79.143.80.210:3000/

lincomatic, 2018-02-09:
Thanks. Turned out I fried my ESP board. It's working now.

OK, so how about I go with the easy idea of red display and a message that stays on the screen until the EV is unplugged?

What should it say? I can't think of anything that really sounds good and fits

1234567890123456
OVER TEMP
LAST CHARGE

glynhudson, 2018-02-09:
Ah excellent. Nice work 👍 

How about high temp, since technically its not yet 'over temp' until the 71 deg C cut off is reached. 

How about just printing a 'HIGH TEMP' warning message to the 1st line of the LCD. That way the info on the 2nd line e.g session energy is not lost. The user will then know how much energy has successfully been put into the car. This could be useful since a charge rate throttle due to over temp may result in a less then required charge. The LCD should also stay red until the car is unplugged. 

Pressing the button or un-plugging the car should clear the message and restore the LCD to normal colour.

lincomatic, 2018-02-09:
I'm a bit puzzled about your statement that the LCD is useful to tell if the EV got less than a full charge. Does anyone actually looks at the EVSE instead of the car to check charge level?

chris1howell, 2018-02-09:
The vehicle could throttle down and still get a full charge before the user drives away. The existing warning would be gone once the station cools down a bit.

The issue that started this discussion was a user had a Over Temperature with some melting of the conductors. After review of the data it was clear that the station had been getting gradually worse over a couple months and was throttling daily. The user never knew he was throttled every night because he got a full charge by morning.

Keeping the warning persistent will allow the user to catch the issue long before a full shutdown.

glynhudson, 2018-02-09:
> Does anyone actually looks at the EVSE instead of the car to check charge level?

I do, I'm sure other (non tesla) nissan Leaf EV drives do since the nissan app is terrible. It takes about 5min to open! I often glace at the LCD when I unplug the car in the morning to check how many kwh has been put into the car. I agree, if had a Tesla with an excellent mobile app this would be the best way to check the charge state. Either way the user has to physically unplug the car at some point therefore they should notice an error on the LCD. 

> Keeping the warning persistent will allow the user to catch the issue long before a full shutdown.

Yes, this is exactly what I'm proposing.

lincomatic, 2018-02-09:
Interesting, I have a Leaf, too. I just check the charge level from inside the car. But on the other hand, I have a LeafCAN on the dash, so I can quickly see it w/o starting the car.

Anyway, I have implemented code to do as we discussed above. Actually, it will also say HIGH TEMP and turn the display red during a current throttling event, even if it's charging. 

Wow, we're nearing filling up our resources again:

Sketch uses 31520 bytes (96%) of program storage space. Maximum is 32768 bytes.
Global variables use 1538 bytes (75%) of dynamic memory, leaving 510 bytes for local variables

I have to be extremely selective about future additions to the code.

lincomatic, 2018-02-09:
I wasn't able to test thoroughly, but it appears to do what we want:

https://github.com/lincomatic/open_evse/commit/c09dea036fa52f6ed5cdd3a132eb7319b654dd4e

closed 2018-02-09 by lincomatic

glynhudson, 2018-02-10:
Excellent! Compiled great, will test tomorrow.

glynhudson, 2018-02-10:
> c09dea0

Your leaf can looks cool BTW! Nice work, just been reading through your
blog archives. I use leaf spy but car needs to be on and phone connected
I.e not fast! Maybe I'll try and build up a leaf can if I have some free
time...

- sent from my mobile device

glynhudson, 2018-02-10:
Just tested, worked exactly as advertised.
 
When a charge is active and high temp is detected LCD flashes red and displayed the current temp and current (throttled) charge rate. Note my testing was done on a test rig hence no current flowing. 

[image]

When the charge finishes "high temp" message is displayed together with the session energy:

[image]