lincomatic/open_evse #83: 230v/400v single/three phase detection and correct energy calculations.

beikeland, 2018-02-21:
I've recently installed an OpenEVSE on a 400V-TN circuit and this allows the connected car to either charge 400V three phase or 230V single phase and it makes the energy counter go out of whack.

Given the OpenEVSE board needs to be wired between L1 and N I take it only knows about 230V regardless of being in a 400V system. Adding a voltmeter wouldn't help either as it wouldn't tell us if the connected car is using L1/L2/L3/N or just L1/N.

Would it be doable to add a a second ammeter as a digital input, monitoring L2 or L3, and based on this decide to calculate the energy based on the correct voltage and correct number of phases? Or is this just something we'll have to wait for the next generation to see resolved?

As a dirty fix I scaled up the current measurements so the 400v/three phase measurements that is what the user is using is correct, but it'll be wrong when some visit with a car that cannot charge off 400V and reverts to pulling single phase 230V from L1 and N.

lincomatic, 2018-02-21:
You need an analog pin for the ammeter. The circuit is actually pretty simple. You could clone it from the OpenEVSE schematic. You can use the pin assigned to VOLTMETER. Then you just need to modify the code to read both ammeters.

lincomatic, 2018-02-21:
This kind of question should be asked on the OpenEVSE google forums, not the github firmware repo

closed 2018-02-21 by lincomatic