lincomatic/open_evse #86: PP_TABLE_TESLA?

glynhudson, 2018-03-29:
Sorry, this is a question rather than an issue. 

It's fantastic that openevse has implemented `PP_AUTO_AMPACITY` for non-tethered EU type-2 socket units. Nice work 👍  

The `PP_TABLE_IEC` resistor and current values all match the EN 61851:1 spec. My question is why the need for `PP_TABLE_TESLA`

https://github.com/openenergymonitor/open_evse/blob/master/firmware/open_evse/AutoCurrentCapacityController.cpp#L30

Tesla's charge using standard EV cables in the EU which have the same PP > GND resistor in the type-2 plug to identify the cables current rating. Why the need for a separate lookup table? I can't see this documented anywhere. 

Thanks.

glynhudson, 2018-04-04:
I guess this must be it. I'm pretty sure that all Teslas (at least in UK/EU) are IEC compliant. Can you remember who contributed this code?

lincomatic, 2018-04-04:
They came from the person who requested the PP support to be implemented. He has since stopped using it, himself. I've removed it to prevent confusion.

https://github.com/lincomatic/open_evse/commit/7f772c469f77199e35cf271f13c1ec9448ed69e4