lincomatic/open_evse #92: Pin change RelayAC

DeltaVetal26, 2018-06-22:
Hi, lincomatic! I need to change pin relayAC from digital to analog. In open_evse.h I found the lines: 
`#define CHARGINGAC_REG & PINB
#define CHARGINGAC_IDX 1`
Everything was fine, if I used a digital pin for the relay, but I only have an analog one. Any ideas? In J1772EvseController.cpp, I found the code responsible for setting the relay pin: 
`#ifdef CHARGINGAC_REG
  pinChargingAC.init (CHARGINGAC_REG, CHARGINGAC_IDX, DigitalPin :: OUT);
#endif`
It's unclear how to change the pin from digital to analog. Any ideas? I apologize for diverting stupid questions.

DeltaVetal26, 2018-06-22:
I went into open_evse.ino and added
 `pinMode(A6, OUTUT)` to the void setup;
Further in J1772EvseController.cpp I'm under each pinChargingAC.write (1);
make `digitalWrite (A6, HIGH);`
And LOW, when charging stops. It is safe?

lincomatic, 2018-06-27:
Sorry for taking so long to get back to you.. been out of town. Which MCU are you using m328P? is it an OpenEVSE board or Arduino? I can't tell you what to do unless I know what hardware you're using.

DeltaVetal26, 2018-06-28:
I use arduino m328p board. Simply, the relay contact is connected to the analog contact, and not to the digital one, to which the relay was connected in the usual OpenEVSE. I did, as I wrote in the previous post, the system works fine, the relay turns on when charging starts and turns off at the end. But I was not sure of the correctness of this decision.

lincomatic, 2018-06-28:
The way you are doing it will work OK, however, the correct way is to simply change the definition in open_evse.h. Which Arduino board are you using? The ones I use only have A0-A5, no A6. As an example, let's use a Duemilanove board.

A5 = PC5

so all you need to do is change

#define CHARGINGAC_REG &PINC
#define CHARGINGAC_IDX 5

closed 2018-06-30 by lincomatic

DeltaVetal26, 2018-06-30:
@lincomatic 
It works! Thank you very much!