lincomatic/open_evse #97: PCF8574 support

martiinezz, 2018-11-02:
Hi,

I've been trying to compile firmware with PCF8574 support without luck. I have monochrome based LCD. 
So far I did get some garbage on the screen while display works great with this library:
http://www.mathertel.de/Arduino/LiquidCrystal_PCF8574.aspx

What I did in the code:
(openevse.ino)
// if using I2CLCD_PCF8574 uncomment below line  and comment out LiquidTWI2.h above
#include "./LiquidCrystal_I2C.h"

I don't see  LiquidTWI2.h anymore. Just doing this didn't help - display acted like dead.
So next I did 
(openevse.h)

// Support PCF8574* based I2C backpack using F. Malpartida's library
// https://bitbucket.org/fmalpartida/new-liquidcrystal/downloads
#define I2CLCD_PCF8574

No luck as well.

I get garbage (and it's dynamic I can see data changes on the display) only when I do: 

#define I2LCD
but on the other hand there is

#ifndef I2CLCD
#undef I2CLCD_PCF8574
#endif

I have tried several combinations without luck.

I have also tried to replace LiquidCrystal_I2C.h with LiquidCrystal_PCF8574.h but it seems wire and twi libraries are tied with LiquidCrystal_I2C and it's beyond my skills to fix that :(

lincomatic, 2018-11-05:
Hi, make sure you are using the development branch.
please try the following, in open_evse.h:

1. keep I2CLCD enabled
2. enable I2CLCD_PCF8574
3. disable RGBLCD
4. look for the following lines in open_evse.h and make sure LCD_I2C_ADDR is set to the correct I2C address for your LCD

martiinezz, 2018-11-06:
Ok, my bad. In the meantime I've build Barbouri THT version and it works fine with your hints. Turned out that by mistake I soldered 24 MHz crystal onto SMD version hence the garbage on the screen. Now what I get is vent error and I don't know how to wire button as PCF expander 8 pins are all used up by LCD.

lincomatic, 2018-11-06:
Glad you go it working. You don't have to use I2C for the button. Just look open_evse.h I checked in yesterday. There are instructions for using a SPST-NO button on any available ATmega328P I/O pin. Just set BTN_REG/BTN_IDX properly and connect the switch between the I/O pin and GND.

closed 2018-11-06 by lincomatic

vegasbrad, 2020-12-02:
martiinezz, did you figure out the button pin input, I run a PCF8574 based LCD on some of my builds, and use one of the 6-pin ISP header pins for the button input.