lincomatic/open_evse #98: Is it possible to run OpenEVSE without an LCD display?

DeltaVetal26, 2019-01-29:
Hi, lincomatic!

I want to use OpenEVSE firmware without using an LCD screen. As far as I understand, if you simply disconnect the screen from the controller, charging will not start at all. Is there any way around this? All data on the state of charge I get through Wifi (ESP8266).
I tried to disable I2CLCD and other screen settings in open_evse.h, but this leads to errors.

lincomatic, 2019-01-30:
Sorry, there have been tons of changes in the code over the years, and I can't possibly maintain every possible configuration. I just fixed the compile error in the development branch when RGBLCD is undefined. see if that works for you. also, look for "obsolete LED pin"  in open_evse.h .. if you properly hook up green & red status LED's and properly set the defines, they can be used for visual status of the EVSE. Also, a BLUE led used to be connected to the charging pin in older revs of the hardware

DeltaVetal26, 2019-01-31:
@lincomatic 

That helped! Thank you very much!

lincomatic, 2019-02-01:
You're welcome!

closed 2019-02-01 by lincomatic

glynhudson, 2019-02-04:
I can confirm that open_evse can be made to run without an LCD. I've been
running one for a while now using an RGB LED as status.

martiinezz, 2019-07-10:
Glyn - did you do any changes to the code? How did you connect RGB diode?

lincomatic, 2019-07-10:
There are 2 ways to use RGB diodes. The OpenEVSE store carries RGB diodes that are I2C compatible and are drop in compatible. Another way is to find some free pins, and set GREEN_LED_REG and RED_LED_REG in open_evse.h. The blue channel goes to the charging pin.

martiinezz, 2019-07-11:
Do you mean CHARGING_IDX as the charging pin?

lincomatic, 2019-07-15:
> Do you mean CHARGING_IDX as the charging pin?

Yes, the blue LED is driven by the charging pin

martiinezz, 2019-07-16:
Thanks - will try that soon :)