lincomatic/open_evse #100: RAPI command to set voltage

chris1howell, 2019-02-26:
RAPI command to set voltage for power calculations. 

Power calculations for L2 are based on a fixed 240v. Depending on location the baseline voltage could be 208, 220, or 230. 

L1 is based on a fixed 120v which could be 100, 110 or 120.

Accuracy could be greatly improved by sending regular RAPI updates from an external measurement source or a user could set the voltage to match their readings under load at the time of charge.

lincomatic, 2019-03-01:
I guess it will help somewhat, but it's still a hack, since the voltage fluctuates. Best would be to provide an actual voltmeter, such as in the existing OPENEVSE_2 code.
I could add something, unfortunately, I'm swamped right now and won't be able to work on it in the near future

lincomatic, 2019-06-27:
Do we still need this? I doubt I can add it w/o running out of space, now that I've added hooks for ocpp support.. though we will have the space if some features, such as delaytimer or menus are turned off. I still think this is strictly a band-aid, though.. a real voltmeter is what we really need. I get some serious voltage sag in my house as the EV pulls more current... it varies in realtime.

landswellsong, 2019-07-17:
Sorry for interjecting, but how does the idea of moving voltage/current measurements out to a separate module on, say, I²C sound? From a EU perspective, 3×V + 3×I = 6 extra ADC pins are definitely a no-go, but it can be easily offloaded on a separate board.

lincomatic, 2019-07-28:
> Sorry for interjecting, but how does the idea of moving voltage/current measurements out to a separate module on, say, I²C sound? From a EU perspective, 3×V + 3×I = 6 extra ADC pins are definitely a no-go, but it can be easily offloaded on a separate board. Also I imagine it would free quite a bit of program memory by removing the current calculation routine.

That is a good idea. However, you should bring that up in the OpenEVSE hardware forum. This repo is just for firmware

closed 2019-07-28 by lincomatic

landswellsong, 2019-07-29:
@lincomatic thanks will do