lincomatic/open_evse #102: Going to sleep while charging trips GFCI

Stef-Sijben, 2019-04-10:
I've been charging my Renault Zoe with OpenEVSE for about a year without issue. Lately I added a WiFi module and so I now have the option to stop charging by pressing the pause button. Doing this while the car is charging and pulling current often trips the GFCI breaker in my house's power board, which happens at 30 mA residual current.
My set up is as follows (all 3 phase + N devices), which is very common at least in the Netherlands:

`Grid ==> main breaker ==> utility meter ==> GFCI ==> circuit breaker ==> dedicated energy meter ==> relay ==> car`

My hypothesis is that when OpenEVSE signals the car to stop charging via the control pilot, the car immediately changes its pilot status to state B and the EVSE switches the relay off right away. There is no noticeable delay between me pushing the button in the web interface and hearing the relay click. Presumably some residual current is still flowing at that time, which has nowhere to go but the PE lead, thus tripping the GFCI.
Adding a small delay between the the detection of state B and switching the relay off solves the problem for me. I set the delay to 200ms, but I didn't test if smaller values work as well.

I made the following change in J1772EvseController.cpp, starting from line 1101, to insert the delay. Obviously this is an ugly hack, but maybe you could consider building in a good solution to achieve this.

```C
...
    if (chargingIsOn()) {
      // wait for pilot voltage to go > STATE C. This will happen if
      // a) EV reacts and goes back to state B (opens its contacts)
      // b) user pulls out the charge connector
      // if it doesn't happen within 3 sec, we'll just open our relay anyway
      // c) no current draw means EV opened its contacts even if it stays in STATE C
      //    allow 3A slop for ammeter inaccuracy
      if ((phigh >= m_ThreshData.m_ThreshBC)
#ifdef AMMETER
	  || (m_AmmeterReading <= 3000)
#endif // AMMETER
    ) {
      // Pull time-out forward to 200 ms after first detection.
      if (m_ChargeOffTimeMS > curms - 2800) {
        m_ChargeOffTimeMS = curms - 2800;
      }
    }
    if ((curms - m_ChargeOffTimeMS) >= 3000) {
      chargingOff();
...
```

lincomatic, 2019-06-26:
How about if I wait for STATE B && <=3A instead of STATE B || <=3A ?

```
      if (((phigh >= m_ThreshData.m_ThreshBC)
#ifdef AMMETER
	  && (m_AmmeterReading <= 3000)
#endif // AMMETER
	  ) || ((curms - m_ChargeOffTimeMS) >= 3000)) {
	chargingOff();
```

BTW, why do have OpenEVSE connected to a GFCI, since it already has one built-in?

glynhudson, 2019-08-19:
This is a nice addition. I've not heard reports of this issue, however it would be better to wait for the car to fully reduce the current before opening the contactor. I can't see any harm in adding a delay.

>BTW, why do have OpenEVSE connected to a GFCI, since it already has one built-in?

It's an electrical requirement in Europe.

@Stef-Sijben what RCD are you using? EVSE's should have a type-B RCD which is tolerant to DC leakage. However in this case it sounds like it would still trip e.g https://shop.openenergymonitor.com/type-b-rcd-1p-n-chint-nl210-63-263-30/

Stef-Sijben, 2019-08-21:
> How about if I wait for STATE B && <=3A instead of STATE B || <=3A ?

That won't work with my current setup, as I don't have a current transformer installed. Instead I have a separate 3 phase meter that I read via ModBus. I could of course add one if this is the solution you're going with.

> BTW, why do have OpenEVSE connected to a GFCI, since it already has one built-in?

As said, it's required over here, and since I already have that, I haven't installed a built-in one.

> @Stef-Sijben what RCD are you using? EVSE's should have a type-B RCD which is tolerant to DC leakage. However in this case it sounds like it would still trip e.g https://shop.openenergymonitor.com/type-b-rcd-1p-n-chint-nl210-63-263-30/

Actually a type B RCD is **sensitive** to DC leakage in the sense that it trips when more than 6 mA of DC current is leaking, where a regular type A is not required to trip in that case. So a type B should trip in more situations than a type A RCD.

glynhudson, 2019-08-21:
> I can't see any harm in adding a delay.

Actually giving more thought to this, if a delay is to be added it should only be effect when the button is pressed. Not when the EV is disconnected since there is a safety requirement for EVSE to open contactor within 100ms when requested

lincomatic, 2019-12-16:
OK, I'm not going to implement this, since it doesn't seem to be a common problem, and quick response is of utmost importance

closed 2019-12-16 by lincomatic