lincomatic/open_evse #105: Newbie

jbakuwel, 2019-08-19:
Hi folks,

Thanks for all the hard work done on this project, it's much appreciated. And (almost) works like a charm on my off grid system here. I realise I am probably a bit of an exception charging an old 2011 Nissan Leaf on an off grid power system with a few relatively big other power consumers.

One thing that is not working for me is the current over-current defaults in the firmware as it seems the charger in my old Leaf is not responding fast enough to a change in current.

So ... I'd like to compile a custom version of the firmware with the OVERCURRENT_TIMEOUT changed to (an arbitrary) 8000L instead of the default 5000L to see if that is a better fit for my car.

I found open_evse firmware in the OpenEVSE repository as well as here. Unfortunately I can't compile the v5.0.1 firmware as published in the OpenEVSE repository with v1.6.15 of the build tools as that results in a sketch that is too big.

I've tried compiling the development  branch from this repository with RAPI_WF disabled/commented out but that too results in a sketch that's too big. The stable branch does compile but as far as I can tell that is v3.11.3, much older, I presume, than v5.0.1

I'm a bit confused as to which repository I can or should use. The v5.0.1 firmware appears to be pretty good apart from the issue with the default OVERCURRENT values so I'd prefer to first compile v5.0.1 as-is, which should result in the same firmware as I've got running currently. Then I hope I can download the firmware as it is shipped in the OpenEVSE charge controllers so I can confirm that I have been able to compile the exact firmware as that was shipped. Once that all checks out, I can make the one small change and compile the custom firmware.

A few pointers from someone how I can upload that firmware to the OpenEVSE charge controller would be much appreciated. I've done some Arduino stuff in the past but could always upload that straight from the IDE via USB.

And once all this is done, I can start learning more and hopefully one day make more complex and meaningful contributions to the project.

thanks in advance,
Jan

glynhudson, 2019-08-19:
See attached for V5.1.0.EU compiled with `OVERCURRENT_TIMEOUT 8000L`
[firmware.zip](https://github.com/lincomatic/open_evse/files/3516608/firmware.zip)

I maintain the EU version of the open_evse firmware: https://github.com/openenergymonitor/open_evse

The main differance is current is limited to 32A max and L2 is enabled by default. You can manually select L1 if required. I use platfromio to compile, see instructions on the repo. However, you should just be able to load the firmware directly using an ISP programmer: https://openevse.dozuki.com/Guide/How+to+Load+OpenEVSE+Firmware+(WinAVR)/7

This overcurrent issue was fixed in V5.0.1 https://github.com/lincomatic/open_evse/issues/95

Where did you obtain your OpenEVSE controller? What firmware was pre-loaded?

jbakuwel, 2019-08-19:
Thanks for the V5.1.0.EU firmware and link to details re programming.

I bought the OpenEVSE controller from https://openevse.com. It was loaded with firmware v5.0.1 and wifi firmware v2.8.0. I've updated the wifi firmware to v2.8.1 not so long ago.

jbakuwel, 2019-08-19:
Ooops - clicked too fast. One last question :-)

Which repo should I use to update the open_evse firmware on the controller bought from https://openevse.com ? Am I correct in assuming that [this repo](https://github.com/OpenEVSE/open_evse) is no longer in active development? The stable branch in [this repo](https://github.com/lincomatic/open_evse) looks like in active development and [the openenergymonitor](https://github.com/openenergymonitor/open_evse) repo as well. However I can't compile the firmware (sketch too big) from the lincomatic repo (haven't tried the openenergymonitor repo yet). 

Is there a reason why development is not taking place in one repo and branches created for different flavors instead of having multiple repos?

glynhudson, 2019-08-20:
Main Development repo (not for production, could be untested) : https://github.com/lincomatic/open_evse/tree/development

Stable production maintained by OpenEnergyMonitor for EU units sold via OpenEnergyMonitor store: https://github.com/openenergymonitor/open_evse

Stable production maintained by OpenEVSE for units sold via OpenEVSE store: https://github.com/openenergymonitor/open_evse

jbakuwel, 2019-08-20:
Hi Glyn,

Thanks for all your help, it's much appreciated!
We're all biased one way or another :-) As newbie I gladly take advise from folks who've been doing this for a bit longer. I'm not used to Platform IO and find the interface a bit busy and confusing but that's just a matter of getting used to it and will give it another go. I suppose I should be able to compile the firmware from the OpenEnergyMonitor repo with both Platform IO and the Arduino IDE?

Re repo's, I'm in NZ and so it makes sense to use the repo with the EU defaults instead of the US defaults. Besides I can't compile the firmware from the OpenEVSE repo using the available instructions. Hopefully I will be able to do so using the OpenEnergyMonitor store.

closed 2019-08-20 by jbakuwel