lincomatic/open_evse #106: Compiling openenergeymonitor/open_evse

jbakuwel, 2019-08-25:
Hi Glyn,

I do not understand why the compiler issues this warning. It is as if the compiler doesn't know about the #if directive? Any suggestions? I'm compiling the master branch, no local changes, with Platform IO.

While on topic of compiler directives and conditionals, is there a way to test whether we're compiling with Platform IO so we can make this conditional, ie. include the define for the Arduinio IDE but not for Platform IO? Something like:

#ifdef PIO
// #define VERSION "D5.1.0" -- see platformio.ini
#else
#define VERSION "D5.1.0"
#endif

and add "-D PIO" to the src_build_flags in platformio.ini

Jan

/home/dev/OpenEVSE/eu/open_evse/firmware/open_evse/open_evse.ino:1187:0: warning: "SVC_LVL_MNU_ITEMCNT" redefined
 #define SVC_LVL_MNU_ITEMCNT 2
 ^
/home/dev/OpenEVSE/eu/open_evse/firmware/open_evse/open_evse.ino:1185:0: note: this is the location of the previous definition
 #define SVC_LVL_MNU_ITEMCNT 3
 ^

The corresponding section in open_evse.ino is this:

#if defined(ADVPWR) && defined(AUTOSVCLEVEL)
#define SVC_LVL_MNU_ITEMCNT 3
#else
#define SVC_LVL_MNU_ITEMCNT 2
#endif // ADVPWR && AUTOSVCLEVEL

jbakuwel, 2019-08-28:
@lincomatic: please correct me if I'm wrong but as far as I can tell this is the only repo where I can actually post something.

lincomatic, 2019-08-29:
> @lincomatic: please correct me if I'm wrong but as far as I can tell this is the only repo where I can actually post something.

Really? That's strange. Usually, you can post issues to any repo.

OK, I just went over to https://github.com/openenergymonitor/open_evse, and issues are disabled there. I've never seen that. Perhaps you can request them to open up that capability. Thanks.

lincomatic, 2019-08-29:
In the meantime, I will reopen this case for you, in the hopes that @glynhudson or @jeremypoulter responds to you, as I am not qualified to answer it.

jeremypoulter, 2019-08-29:
In answer to the original question...

```
/home/dev/OpenEVSE/eu/open_evse/firmware/open_evse/open_evse.ino:1187:0: warning: "SVC_LVL_MNU_ITEMCNT" redefined
#define SVC_LVL_MNU_ITEMCNT 2
^
/home/dev/OpenEVSE/eu/open_evse/firmware/open_evse/open_evse.ino:1185:0: note: this is the location of the previous definition
#define SVC_LVL_MNU_ITEMCNT 3
^
```
This is from the .ino to .cpp conversion guess that does not like this, but the actual compilation handles it fine.

> While on topic of compiler directives and conditionals, is there a way to test whether we're compiling with Platform IO so we can make this conditional, ie. include the define for the Arduinio IDE but not for Platform IO? Something like:

```c
#ifdef PIO
// #define VERSION "D5.1.0" -- see platformio.ini
#else
#define VERSION "D5.1.0"
#endif
```

Probably not a good idea to start doing if PlatformIO do this if Arduino do that, however if there is a genuine reason it would be better to do something like (IMHO);
```c
#ifndef VERSION
#define VERSION "D5.1.0"
#endif
```

lincomatic, 2019-10-04:
Why are you moving the version number to an ini file? It should be defined in the code.

jeremypoulter, 2019-10-05:
Personally I am not proposing that, I think it was just an example. That being said there is lots of cool stuff you can do if the VERSION is able to be overridden, eg with Travis CI you could just tag in Git and automatically pull the version from the tag, build all the different variants, and upload to the Git releases page.

lincomatic, 2019-10-05:
OK, good idea. It is a great way to make custom builds. It's easy enough to change. I guess all of the standard features should be defined that way, too.  I'll put it in the next commit

lincomatic, 2019-12-18:
the #ifndef around VERSION has been pushed.

closed 2019-12-18 by lincomatic