lincomatic/open_evse #111: Physical Security - RAPI command to disable button.

chris1howell, 2019-12-03:
RAPI command needed for physical security to disable push button.

lincomatic, 2019-12-13:
no code space left to implement this, unfortunately.

beikeland, 2019-12-16:
Maybe I'm missing the use case, but if the button is installed, I presume its meant for user access.

Currently its not possible to unlock AuthLock from the menu, as such if the code is made to ignore the button when AuthLock is enabled then only authorized users will have access to use the button.

If authorized users aren't supposed to access the button then I don't see the need to install/connect it.

Should it be neccecary to disable the button it could be handled by the RAPI client by connecting one side of the button to say the ESP8266 or DTR line if its a serial server etc.

lincomatic, 2019-12-17:
authlock prevents unauthorized charging. button locking prevents unauthorized changing of settings. like for instance, max amps, and even scarier, turning off safety checks

beikeland, 2019-12-31:
I get that, but if someone authorized to charge is not allowed to access the menu, and RAPI is required to enable the button then whoever can enable the button can also make the desired changes directly over RAPI which to me is faster and easier. So if being reliant on RAPI is acceptable the button could be skipped or the menu stripped down without any real loss of functionality, freeing up space for the needed RAPI command.

craigk46, 2019-12-31:
Bjørn,
You need to be clear about what your desired behavior is that you request.  My head spins a little bit reading and re-reading what you wrote and trying to figure out what you meant, what you seek.

No matter.

Some advice:
If you have an OpenEVSE with no button you can make free >10Kbytes of eprom by commenting out #define BTN_MENU in open_evse.h and it will make the button still useful for pausing during charging but just remove the ability to access the menus.
If you want to “trim” the menus a little bit but not completely remove the menus you can add #undef DELAYTIMER_MENU  right after you see #define DELAYTIMER_MENU also in open_evse.h  This saves 2,782 bytes and still the timers are functional and can be set over WiFi.
Sam is correct that it would be best to have a new RAPI command for disabling the button.  I played a code change disabling the button during AUTH_LOCK and it does work nicely but the button remains available to tamper during charging.  If we are seriously concerned about button tampering then a new RAPI button disable would be a nice enhancement.  A fine alternative is to build the kit with the button disconnected as long as WiFi is okay.

Keep having fun,
Craig K.

craigk46, 2020-01-02:
Since this thread is concerned with security of the OpenEVSE I submit to you this short video of firmware I developed today.
https://www.dropbox.com/s/d781ln60ea24uar/AB1D201F-97F5-431B-8B6A-73ED3BEAAD36.mov?dl=0 <https://www.dropbox.com/s/d781ln60ea24uar/AB1D201F-97F5-431B-8B6A-73ED3BEAAD36.mov?dl=0>

I did something in this code to disable the button during AUTH_LOCK 1.   A better approach I think is to have what was originally requested which was a RAPI command to disable the button entirely until a RAPI command re-enables the button.  Code space is very limited at this time for such enhancement requests.  Experimenters such as we can disable DELAYTIMER_MENU in order to gain some working space for firmware enhancements.

Keep having fun,
Craig K.

lincomatic, 2020-01-03:
Looks cool, Craig! Is the EVSE managing the NFC board, or an external processor?

craigk46, 2020-01-03:
I used an external Adafruit ProTrinket board (in place of having it controlled by the ESP8266).  This was just to make something quick and dirty as a simple prototype. 

Best Wishes,
Craig K.

Looks cool, Craig! Is the EVSE managing the NFC board, or an external processor?

chris1howell, 2020-03-27:
I would like to expand this a little further to enable an external processor (ESP32 WiFi) take over UI and writing to the LCD. My suggestion is:
-Remove all menus from main OpenEVSE controller except Service Level, Current and GFI self test, delay timer menu code alone should free up enough memory.
-new RAPI command to cause the OpenEVSE controller to ignore button presses BUT pass an RAPI message for short press, long press and 10s press.

lincomatic, 2020-03-28:
https://github.com/lincomatic/open_evse/commit/c226ad80166ba79d47b27a96e911276b0a0cacb6
https://github.com/lincomatic/open_evse/commit/9c487b5acd16dc5aef9e65ce1f4376959e56c95e

closed 2020-03-28 by lincomatic

jeremypoulter, 2020-12-30:
> ... BUT pass an RAPI message for short press, long press and 10s press. 

looks to be missing, was there a reason?