lincomatic/open_evse #118: RAPI command to set MAX current

chris1howell, 2020-03-27:
Setting the maximum current for a station requires compiling the firmware for each common value, which can be challenging. Leaving the Maximum at the J1772 80A covers everyone however it can allow a station to be set above the ability of the hardware.

A RAPI command to set the station maximum current would be very helpful, for security the RAPI command should only allow write to EEPROM once. Changing the value again would require erasing the EEPROM or re-flashing firmware.

lincomatic, 2020-03-27:
I really don't agree w/ loading a firmware into an EVSE that's even capable of exceeding the hardware limit, even if it's a "protected" value. If you want to avoid recompiling, it's not that hard to write a program that edits a binary firmware image and sets the max value before burning the firmware into an EVSE.

lincomatic, 2020-03-27:
In https://github.com/OpenEVSE/openevse_wifi_gui/issues/12, Jeremy mentioned another pressing need, which is to know the maximum current capacity which was loaded in from EEPROM at boot time. I will add a parameter to $GC for this

chris1howell, 2020-03-27:
> In OpenEVSE/openevse_wifi_gui#12

Firmware is loaded at the distributor before the board is even fabricated.
It is not practical to manufacture and stock all possible variations. But
sending an RAPI command or writing EEPROM is doable.

Defaults are 80A max, 24A current GFI self test disabled.

Currently there is no easy way to set absolute max so everything ships 80A.

lincomatic, 2020-03-28:
> In [OpenEVSE/openevse_wifi_gui#12](https://github.com/OpenEVSE/openevse_wifi_gui/issues/12), Jeremy mentioned another pressing need, which is to know the maximum current capacity which was loaded in from EEPROM at boot time. I will add a parameter to $GC for this

https://github.com/lincomatic/open_evse/commit/68f362cd073827eae03f8dace6947072f428e919

lincomatic, 2020-03-28:
> Firmware is loaded at the distributor before the board is even fabricated. It is not practical to manufacture and stock all possible variations. But sending an RAPI command or writing EEPROM is [...]

Running RAPI commands during assembly to do configuration sounds too error-prone, and adds labor during manufacturing, too. I feel pretty strongly that the firmwares should be hardcoded for the proper upper current limit. As I stated above, the best approach is to write a program that edits the compiled firmware file to set whatever upper current limit is desired.

chris1howell, 2020-03-28:
I agree in principal. But in reality not every customer is savy enough to
setup an IDE and reprogram their controller. We can't possibly interview
each customer who orders a controller, kit or station and compile custom
code to match their final setup or release 74 versions of each release (or
even a dozen of the most common).

In reality there are thousands of controllers out there set to 80A because
that is the default that works for everyone. The only workable solution
today is to load common firmware at the distributor with universal settings
(80A max).

Ideally all stations would have the Max set to the stations limit but
almost none are because there is no easy way to do it during final
production or at home.

lincomatic, 2020-04-21:
https://github.com/lincomatic/open_evse/commit/044b62fec924530dbd62e6605fc95f854e216b9c

closed 2020-04-21 by lincomatic