lincomatic/open_evse #121: RAPI command to set voltage

jeremypoulter, 2020-04-29:
For the solar divert we are adding config to allow setting the voltage to either a static value or an MQTT feed from an energy monitor for live updates. This should allow for better conversions from power to amps. 

It would also make sense to also feed this into the EVSE so it can better calculate the power.

lincomatic, 2020-04-30:
Does this need to be a non-volatile value, or can it just reload the default at each boot?

glynhudson, 2020-04-30:
Volatile is fine, and preferable IMO. I think most users using PV divert would choose to have VRMS set via a real time MQTT feed since it's a requirement for PV diver to have an energy monitor which will also be monitoring VRMS

lincomatic, 2020-05-08:
https://github.com/lincomatic/open_evse/commit/b1b60491d38e389c1135d79944d1e163954b8c0d

closed 2020-05-08 by lincomatic

jeffl012, 2020-05-25:
I see this thread is closed, but would it be possible to add a manual setting similar to the "SC amps [V] - set current capacity" command? This way a user could hard set the voltage one time for their environment (ie: 208 or 240) in lieu of obtaining dynamic energy monitor readings and obtain more accurate kWh output.