lincomatic/open_evse #122: Variables undefined if KWH_RECORDING is not defined. Won't compile.

sorphin, 2020-09-26:
Recommend moving definition outside of EnergyMeter.h or outside of ifdef.

**Errror:**
/var/folders/5x/8dx_s69n7xx907bbhsq42hrh0000gn/T/arduino_build_369144/sketch/J1772EvseController.cpp: In member function 'void J1772EVSEController::SetSvcLevel(uint8_t, uint8_t)':
J1772EvseController.cpp:579:17: error: 'MV_FOR_L2' was not declared in this scope
     m_Voltage = MV_FOR_L2;
                 ^~~~~~~~~
J1772EvseController.cpp:584:17: error: 'MV_FOR_L1' was not declared in this scope
     m_Voltage = MV_FOR_L1;
                 ^~~~~~~~~
exit status 1
'MV_FOR_L2' was not declared in this scope

---

**Code:**
void J1772EVSEController::SetSvcLevel(uint8_t svclvl,uint8_t updatelcd)
{
#ifdef SERDBG
  if (SerDbgEnabled()) {
    Serial.print("SetSvcLevel: ");Serial.println((int)svclvl);
  }
#endif //#ifdef SERDBG
  if (svclvl == 2) {
    m_wFlags |= ECF_L2; // set to Level 2
    **m_Voltage = MV_FOR_L2;**
  }
  else {
    svclvl = 1; // force invalid value to L1
    m_wFlags &= ~ECF_L2; // set to Level 1
    **m_Voltage = MV_FOR_L1;**
  }

----

**Defined at beginning of EnergyMeter.h:**
#ifdef KWH_RECORDING

#define MV_FOR_L1 120000L       // conventional for North America
//  #define VOLTS_FOR_L2 230   // conventional for most of the world
#define MV_FOR_L2 240000L       // conventional for North America

lincomatic, 2020-10-01:
https://github.com/lincomatic/open_evse/commit/46c77bea4a1989b8f101a1fcc7d4c49fe8a3151a

closed 2020-10-01 by lincomatic

sorphin, 2020-10-03:
Didn't fix it here because the location you moved it is covered by #ifdef AMMETER, but the compile fault is in below which doesn't require AMMETER.
:
**void J1772EVSEController::SetSvcLevel(uint8_t svclvl,uint8_t updatelcd)**
{
#ifdef SERDBG
  if (SerDbgEnabled()) {
    Serial.print("SetSvcLevel: ");Serial.println((int)svclvl);
  }
#endif //#ifdef SERDBG
  if (svclvl == 2) {
    m_wFlags |= ECF_L2; // set to Level 2
    **m_Voltage = MV_FOR_L2;   <---------**
  }
  else {
    svclvl = 1; // force invalid value to L1
    m_wFlags &= ~ECF_L2; // set to Level 1
    **m_Voltage = MV_FOR_L1;   <----------**
  }

  SaveEvseFlags();

  uint8_t ampacity = GetMaxCurrentCapacity();

  SetCurrentCapacity(ampacity,0,1);

  if (updatelcd) {
    g_OBD.Update(OBD_UPD_FORCE);
  }
}