lincomatic/open_evse #123: Measuring threephase power

faustomilletari, 2020-09-29:
Dear open_evse community. First of all i would like to thank you for your awesome project. really cool.

I was wondering if it would be possible to enable automatic threephase/singlephase power calculation by changing this line 

https://github.com/lincomatic/open_evse/blob/baf49a8d36a9292e1929d3fbe69808d92190d299/firmware/open_evse/EnergyMeter.cpp#L62

to something that checks the optocoupler status by invoking `g_EvseController.ReadACPins()` and check if the status is both low, or only one of them low.

Please let me know your thoughts!

faustomilletari, 2020-09-29:
i guess it should be something along the lines of

```
if (g_EvseController.ReadACPins() == 0)  {
   // three phase
} else {
  // single phase
}
```

i think one could also detect error if the function returns 3?

lincomatic, 2020-09-30:
Sorry the AC pins can't be used to detect 3-phase.

faustomilletari, 2020-10-01:
thank you for your answer. I thought i read in the slides "theory of operation" of open evse hardware that the AC TEST pins, which are connected to a 2-way optocouplers could actually reveal if the power supply was three phase.

I see this here "(1/3 phase detection in Europe)":
[image]

lincomatic, 2020-10-01:
OK, sorry, I was not aware of the usage of the pins for 1/3 phase detection. As you can see from the truth table in your slide above, there isn't any combination shown which specifies what to expect. I am guessing that if one makes the assumption that L1 is not a possibility, that there is a way to deduce 1/3 phase. However, I am not knowledgeable enough to know how to do this. Perhaps we can get @chris1howell to chime in and clarify.

faustomilletari, 2020-10-01:
Fist of all, thank you very much. This project is the most inspiring thing i have come across since long time.

I guess that this way of detecting three phase is only working well in Europe where we don't really have 2 phase systems. 

I will try this out and let you know if it works. I don't think it should go in the official code base tho, because for north america it would not work well. Maybe if you are making a distrinction in the firmware based on region, we don't really have 2 phase in eu, so in theory it could be a new feature to add. 

many thanks!!

chris1howell, 2020-10-01:
OpenEnergyMonitor in the UK is maintaining a EU fork of OpenEVSE for both
Single and Three phase.

https://github.com/openenergymonitor/open_evse

faustomilletari, 2020-10-01:
i think we can close the issue. thanks chris!

closed 2020-10-01 by faustomilletari

beikeland, 2020-12-08:
Said fork doesn't autodetect single or three phase, its simply has an option of applying *sqrt(3) at compile time, and assumes all loads to be three phase if enabled.

I suppose you could detect three phase voltage by the same logic as split phase, but that only determines the supply not the load, to be able to get the power calculation correct you'd need to monitor the current of each phase and determine if the load was on one, two or three phases. I only added one more CT and ignore two phase charging as its quite rare (only european teslas perhaps?)