lincomatic/open_evse #136: Connected state change while sleeping

jeremypoulter, 2021-01-09:
Sure I checked it when fixed, but #115 looks to have returned. I had a quick look and https://github.com/lincomatic/open_evse/blob/development/firmware/open_evse/rapi_proc.cpp#L1023 is checking that the EVSE state has changed, which it does not because it we are sleeping. I added a quick hack, #135, to disable `nodupe` when sleeping, but sure there is going to be a much better way to fix.

Additionally even with this 'fix' the pilot state is not updated and remains the state before going to sleep. The vflags however also gives me the connected state so not a big deal.

lincomatic, 2021-01-12:
D7.1.1 should fix this issue for you. I modified the code to send $AT when any of ECVF_CHANGED_TEST bits change

jeremypoulter, 2021-01-12:
Thanks, I will test tomorrow

jeremypoulter, 2021-01-24:
Looks to be working, sorry I forgot to comment sooner.

closed 2021-01-24 by jeremypoulter