lincomatic/open_evse #139: EVSE reboot when going into an error state

jeremypoulter, 2021-01-24:
In the latest firmware the OpenEVSE will reboot a few seconds after going into an error state, see https://photos.app.goo.gl/7H7WNYdEigqJ2zjL7 

FYI the serial log is 
```
$FP 0 1 Lifetime 0kWh^64
$OK^20
$FP 13 1       ^21
$OK^20
$AT 05 05 32 0702^35
$GF^25
$OK 0 0 0^30
$FB 1^31
$OK^20
$FP 0 0 VEHICLE ERROR^32
$NK^21
$FP 13 0       ^20
$NK^21
$FP 0 0 VEHICLE CHECK^2C
$NK^21
$FP 13 0       ^20
$NK^21
$GF^25
$OK 0 0 0^30
$AT 00 00 0 0200^03
$GS^30
$AB 00 D7.1.1^54
$OK 00 0 00 0300^13
$GP^33
$OK -2560 -2560 -2560^2C
$AT 02 02 32 0700^37
$FF B 0^56
$OK^20
$F0 0^42
$OK^20
$FB 3^33
$OK^20
$FP 0 0 Connected 0A^1C
$OK^20
$FP 12 0       ^21
$OK^20
$GU^36
$OK 0 42^16
$FP 0 1 Lifetime 0kWh^64
$OK^20
$FP 13 1       ^21
$OK^20
$AT 05 05 32 0702^35
$GF^25
$OK 0 0 0^30
$FB 1^31
$OK^20
$FP 0 0 VEHICLE ERROR^32
$NK^21
$FP 13 0       ^20
$NK^21
$FP 0 0 VEHICLE CHECK^2C
$NK^21
$FP 13 0       ^20
$NK^21
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 02 32 0700^37
$FF B 0^56
$OK^20
$F0 0^42
$OK^20
$FB 3^33
$OK^20
$FP 0 0 Connected 0A^1C
$OK^20
$FP 12 0       ^21
```

lincomatic, 2021-01-26:
Doesn't happen to me.

$FR
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 03 04 12 0541^35
$AT 05 05 12 0503^34

https://youtu.be/Em3vQqQMcVw

jeremypoulter, 2021-01-26:
Maybe it is one of the RAPI commands... I will try and do some more testing but is very reprodusable for me...

lincomatic, 2021-01-26:
Maybe it's your compiler? Try my HEX file.

[D7.1.1.hex.zip](https://github.com/lincomatic/open_evse/files/5870610/D7.1.1.hex.zip)

jeremypoulter, 2021-01-28:
Seeing the same issue with that binary

jeremypoulter, 2021-01-28:
Here you go, found the what should be a reliable way to reproduce the error, looks to be related to disabling the front button:

```
jeremy@frenzy:~/Dev/JeremyPoulter/open_evse$ avrdude -e -v -p atmega328p -C /home/jeremy/.platformio/packages/tool-avrdude/avrdude.conf -c usbasp -D -U flash:w:D7.1.1.hex:i

avrdude: Version 6.3
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/jeremy/.platformio/packages/tool-avrdude/avrdude.conf"
         User configuration file is "/home/jeremy/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as FD
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "D7.1.1.hex"
avrdude: writing flash (29710 bytes):

Writing | ################################################## | 100% 10.81s

avrdude: 29710 bytes of flash written
avrdude: verifying flash memory against D7.1.1.hex:
avrdude: load data flash data from input file D7.1.1.hex:
avrdude: input file D7.1.1.hex contains 29710 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.57s

avrdude: verifying ...
avrdude: 29710 bytes of flash verified

avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:DF, L:FF)

avrdude done.  Thank you.

jeremy@frenzy:~/Dev/JeremyPoulter/open_evse$ pio device monitor --baud 115200 --echo --eol CR
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
$GV
$OK D7.1.1 5.1.3^64
$FF G 0
$OK^20
$FF F 0
$OK^20
$FR
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 03 12 0701^35
$AT 05 05 12 0703^36
$FR
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 03 12 0701^35
$AT 05 05 12 0703^36
$FF B 0
$OK^20
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 03 12 0701^35
$AT 05 05 12 0703^36
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 03 12 0701^35
$AT 05 05 12 0703^36
$AT 00 00 0 0200^03
$AB 00 D7.1.1^54
$AT 02 03 12 0701^35
$AT 05 05 12 0703^36

```

https://photos.app.goo.gl/W6CeGgpUwTQWokwn9

glynhudson, 2021-02-03:
I can confirm this behaviour is effecting stable firmware V7.0.2 when front button is disabled

lincomatic, 2021-02-09:
fixed: https://github.com/lincomatic/open_evse/commit/985f682a60fe44ba5e737c03ded3b3f2a9465366
actually, i think it was caused by https://github.com/lincomatic/open_evse/pull/134
the RAPI_BTN code that was added was calling m_Btn.read() w/o patting the dog first, so it was timing out, and causing a reboot.

closed 2021-02-09 by lincomatic