lincomatic/open_evse #140: Random resets after D7.1.3 upgrade

kallisti5, 2021-04-22:
Seeing random "resets" while charging after upgrading my OpenEVSE from 5.0.1 to D7.1.3

I only noticed because I was nearby and head the OpenEVSE contractors open, close in the middle of a charge.    I've heard this happen a few times now while charging my Bolt EV from 52% to 80% at 32A

The only data i've been able to capture so far is a few screenshots of "before and right after" the contactor open.

[image]
[image]
[image]

kallisti5, 2021-04-22:
Hm.  The center of the J1772 is running ~84C while charging.  That's pretty much the top of the spec.   Maybe i'm triggering some thermal shutdown?

lincomatic, 2021-04-23:
84C is very hot. The panic temperature is 71C, and will cause the contactor to open until it cools down. You need to open up your case and figure out where the heat is coming from. You probably have a wire that's not connected properly

closed 2021-04-23 by lincomatic

kallisti5, 2021-05-17:
For any future lurkers.  This is the cable I used:
https://www.amazon.com/gp/product/B07YD2LBZW

You can't open the J1772 connector, it's sealed closed.  I'd avoid connectors of that design at all costs.  I checked the purchase date... it went through minimal abuse (pandemic) for a little over a year before wearing to the point it gets 84C in ~1 hour.

kallisti5, 2021-05-18:
The handle was pretty boring, but actually looked like it was constructed with care. The fault was the design on the pins.  They're solid metal tubes with a web of fine 0.35mm fingers that ride the pin.

After around a year of use, i'm guessing the fingers just "give out" making the few remaining ones touching the pin to heat up.  My other J1772's all have thick 2mm fingers that are wrapped in spring steel to keep them tight.

[image]
[image]
[image]

lincomatic, 2021-05-21:
Yikes, what a piece of junk!