lincomatic/open_evse #142: Connect LED without i2c

DylanVautrin, 2021-07-20:
Hi !
After looking at this issue : lincomatic/open_evse#98
I tried to connect RGBLED to open evse without I2C, so after looking for free pin , i've undefined the RGBLCD to use pin of the LCD in first time, but when i try to write a HIGH/LOW levels on pin SDA, the Open EVSE restart continually.
I used the shortcut "SDA" in the code.
Maybe i do something wrong, someone have a solutions or a different way to proceed ?
Thanks
PS: I just do a digitalWrite(SDA,HIGH);

lincomatic, 2021-07-20:
I'm not sure what you're trying to do. Follow the instructions in the thread #98. Why are writing the SDA pin when you're using an RGB LED? or are you talking about a RGB LED that's connected to I2C? that's a totally different animal. In that case, just leave RGBLCD enabled.

jeremypoulter, 2021-07-21:
There is also this https://store.openevse.com/collections/all-products/products/copy-of-openevse-rgb-lcd or the new WiFi board https://store.openevse.com/collections/all-products/products/openevse-wifi-kit both of which have an LED and temperature on.

DylanVautrin, 2021-07-21:
Thanks for responses.
I'm trying to use RGBLED instead of LCD, but i don't really know which pins use so i try to use the pin for the LCD.
More simple : I just looking for free pin and how connect LED to the Open EVSE.
I've followed the instruction in the thread but it doesn't help me to find the pin physically on the board.
Thx for helping

DylanVautrin, 2021-07-21:
PS: I don't want to use the ESP but the ATMEGA

lincomatic, 2021-07-21:
Ah, OK, if you're using any of the current OpenEVSE boards instead of DIY, then , you're pretty limited. The only unused pins, I can think of are the SCL/SDA/PP pins (as long as you aren't using I2C for anything else). Check the schematic to make sure it doesn't have pull up resistors on SCL/SDA for R/G if you want to go that route. I think the code assumes the blue LED is connected to the charging pin. 
Also, you will need a pin for the button. I guess you can use the PP pin.

Once you have your config settled, I can try to help you figure out the details

lincomatic, 2021-07-21:
BTW, I don't have time to spoon feed you step by step, so you have to have a basic understanding of how to wire up a CC or CA RGB LED and what resistor to use if you want my help.

closed 2021-08-03 by lincomatic