lincomatic/open_evse #144: Automated build and merging forks

jeremypoulter, 2021-11-09:
For the WiFi module project we now have GitHub actions setup to [build](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/actions) 
 and create the [firmware releases](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/latest).

Using GitHub actions makes the release process a lot more robust and repeatable and ensures that the firmware can be built from the source committed to Git.

I would like to setup something similar for the EVSE firmware where we merge all the changes in the OpenEVSE and OpenEnergyMonitor forks back into this repo and then use the PlatformIO envs (via a GitHub Action) to build all of the different configs required as well as generate a pre-release with the built binaries. This can then be manually tested (hopefully we can automate this at some stage) and promoted to a full release when ready.

@lincomatic / @chris1howell / @glynhudson what do you think?

chris1howell, 2021-11-09:
Great work Jeremy. I agree we should do the same thing for the OpenEVSE
firmware.

lincomatic, 2021-11-09:
Sounds like a great idea, but I don't see how this is possible given the brittle nature of Arduino. I am going to stick w/ the UL approved build environment/compiler/library rev. I actually tried building w/ PlatformIO last year, and as I predicted some strange behavior popped up, due to PlatformIO not having an identical toolchain/Arduino library rev available.
It would be better use the OpenEVSE repo as the merge target, rather than mine, because my codebase isn't tailored specifically for the "official" hardware

jeremypoulter, 2021-11-09:
I will have to check again but I think I got it building with the identical compilers as specified, the main reason to use Platform IO it has a lot more support for building different configs. That being said, Arduino has improved a lot in the CLI front so may be worth revisiting building using that directly. 

@chris1howell is doing another round of UL certification so does this give us an opportunity to update the toolchain? 

Fair-enough about the using the OpenEVSE repo for this work, @glynhudson is that ok with you?

lincomatic, 2021-11-10:
Alternatively, we could use different branches, maybe move the release code to the stable branch, and  could rename the current development branch to something else.
I'm OK if you guys prefer to use PIO, as long as the code is set up so it still compiles w/ the Arduino IDE.

jeremypoulter, 2021-11-11:
Let me put together a PR and we can work out where to put it later, not a big deal to change repos/branches

jeremypoulter, 2021-11-11:
Also @lincomatic / @chris1howell can you clarify the relationship between UL certification and the compiler/build env as - this request is ultimately to support that (creates a defined/audited build/release process)

chris1howell, 2021-11-12:
The first UL certification owned by Wattzilla and used the Arduino IDE with
a certain version of the tool chain.

The current UL effort is a clean slate, we can document and define the
process and tools however we like.

Unless Wattzilla updates their data and documentation they will have to
continue to build as approved by their certification.

lincomatic, 2021-11-16:
@jeremypoulter, @chris1howell's summary above is the exact situation. Actually, the only currently UL-approved configuration is with Wattzilla's hardware, which is not exactly the same as a generic OpenEVSE configuration.

lincomatic, 2021-11-16:
@jeremypoulter, the PR you submitted for the automated build process has too many commits, many of which aren't related to the build process. I suggest to either move it to the stable branch, or we can create a new branch for it. Also, the documentation contains references to OpenEnergyMonitor, and I don't think those belong in my repo.

jeremypoulter, 2021-11-16:
That PR is still work in progress at the moment, I will mark as not draft when ready. 

The majority of that is merging the EnonEVSE work upstream so we can all build from the same source. I can split into multiple PRs if you would prefer but let me finish the implementation then we can decide where it can all live.

jeremypoulter, 2021-11-21:
Ok, I think the update are done. @lincomatic @chris1howell @glynhudson can you take a look at #145 ? An example of the (pre-)release can be found <https://github.com/jeremypoulter/open_evse/releases/tag/latest>. Thinking about it some more I think maybe the 'official' releases should be from the OpenEVSE fork, but it would make it much easier to maintain if everything was merged here.

jeremypoulter, 2022-01-04:
These changes have been made in the OpenEVSE fork and this will be the home of the 'official' releases, possibly at some point we should merge the OpenEVSE/OpenEnergyMonitor changes upstream, but this is quite a lot of work to un-pick in to distinct changes

closed 2022-01-04 by jeremypoulter