lincomatic/open_evse #147: RAPI $ST command

lolo1878, 2021-12-31:
Hello,
I would like to use the RAPI $ ST command to configure the working time range, but this command doesn't seem to work.
Can you confirm to me if this command is functional?
Openevse version: 7.1.3.EU
OpenEVSE Wifi version: 4.1.0
cordially
Laurent

chaseadam, 2022-01-05:
I can confirm it works with the following versions: {"cmd":"$GV","ret":"$OK 7.1.3 5.1.3^22"}%       

Note 5.1.3 vs your 4.1.0, so not sure

lolo1878, 2022-01-05:
Thank you for your answer.
indeed when I run the command $ GV I have this result:> $ OK 7.1.3.EU 5.1.3 ^ 1C
I am therefore on the same version as you.
when I issue the command $ ST 22 00 07 00 and press send, nothing happens (no feedback) and the modification is not taken into account.
Maybe I am misusing this command?
can you take the test and give me your result.
thank you very much

chaseadam, 2022-01-05:
should get an "ok"

This is the response from the HTTP rapi call (wifi installed)

http://10.2.0.110/r?json&rapi=$ST+21+0+6+0

```
{"cmd":"$ST 21 0 6 0","ret":"$OK^20"}%
```

lolo1878, 2022-01-06:
this is the reponse:
`{"cmd":"$ST 21 0 6 0","error":"RAPI_RESPONSE_BLOCKED"}`

jeremypoulter, 2022-01-06:
This is the WiFi firmware blocking the timer commands, as of v4 the WiFi firmware implements the timer functionality, not the EVSE module. You can use the [/schedule](https://openevse.stoplight.io/docs/openevse-wifi-v4/b3A6MjA2MTYyNDI-list-schedule-events) API to get/set the timers.

lolo1878, 2022-01-06:
Thanks for your help, but I'm not sure how to get around my problem.
I am a novice and I do not know the /schedule API

jeremypoulter, 2022-01-06:
Sorry, should have been less clever with my markdown. Have a look at https://openevse.stoplight.io/docs/openevse-wifi-v4/b3A6MjA2MTYyNDI-list-schedule-events

lolo1878, 2022-01-06:
I'm not good enough to figure out how I'm going to get the information from your link.
As much as I knew how to easily use the LPR commands, but the wifi api, I don't know.

jeremypoulter, 2022-01-06:
If you look to the right on the documentation there are several examples from various languages, but if the /schedule API is to complex we can look at a simplified version to replicate setting a timer that start/stops the same time on every day.

The other alternative is to use the v3 firmware, that doesn't have the restrictions.

lolo1878, 2022-01-07:
What interests me is to be able to activate / deactivate the programming of the time slot knowing that it will be fixed.
Thank you for your help.

chaseadam, 2022-01-07:
@lolo1878 looks like the method I was using is no longer available. Assuming you are using the wifi module to program there is a newer and better method @jeremypoulter mentioned above. Just upgraded to Wifi version 4.x (which I assume you are using) and it works great.

https://openevse.stoplight.io/docs/openevse-wifi-v4/b3A6MjA2MTYyNDI-list-schedule-events

lincomatic, 2022-01-08:
This is not an issue w/ this repo, so I am closing it.

closed 2022-01-08 by lincomatic