lincomatic/open_evse #148: `ECF_AUTO_SVC_LEVEL_DISABLED` not set when `AUTOSVCLEVEL` is not defined

jeremypoulter, 2022-01-04:
When `AUTOSVCLEVEL` is disabled (as it is in the EU versions of the firmware) the `ECF_AUTO_SVC_LEVEL_DISABLED` flag is not set so the WiFi controller thinks the controller is in auto mode, no matter what is sent via `$SL`.

lincomatic, 2022-01-08:
I had assumed that builds w/ AUTOSVCLEVEL disabled wouldn't need to test the flag, since it's not an option to enable it.
You could just disable the flag with a build that has AUTOSVCLEVEL enabled, save the EEPROM, and burn that saved file into units that need the flag set.
Anyway, it was just a 1-line change. 
https://github.com/lincomatic/open_evse/commit/d9ec7fed68c9ae7fa18d15663d943bb1d5d0d59c

closed 2022-01-08 by lincomatic

jeremypoulter, 2022-01-08:
Thanks, there is not a separate build of the WiFi firmware for US/EU so it has no knowledge of the features enabled in the firmware.