lincomatic/open_evse #151: OpenEVSE state flow chart and testing

jeremypoulter, 2022-01-19:
Looking at a test procedure for OpenEVSE and found this [flow chart](https://openev.freshdesk.com/helpdesk/attachments/6064019242) in the OpenEVSE knowledge base. @lincomatic / @chris1howell can you confirm this is still valid as I think a good place to start would be to validate that the flow chart is followed.

Is there any other existing process and/or documents describing the test procedure?

jeremypoulter, 2022-01-19:
Also assuming the flow chart is correct, why is the ground check dependant on the auto service level? I would assume that this check is disabled when the Auto service level is disabled as with the EU builds (@glynhudson)?

lincomatic, 2022-01-20:
The ground check is done whenever the relay is closed, so it's monitored continuously when the EV is charging. When the new CGMI hardware is released, and CGMI is enabled in the firmware, it will be checked continuously, even when the EV isn't charging.

lincomatic, 2022-01-20:
The only thing I see missing in your flowchart is that if the EV is plugged in during POST, we can't do auto service level detection, so it's done at the beginning of the transition into STATE C, and the max current is adjusted on the fly.

jeremypoulter, 2022-01-21:
Thanks for the info

closed 2022-01-21 by jeremypoulter

lincomatic, 2022-02-06:
I forgot to add 2 more things.:
1) GFCI self test is performed every time a charging is requested (this is why there is always a delay before the contacts close). If it fails, it is a hard fault.
2) any fault which occurs within 2sec of the start of charging is treated as a hard fault
The difference between a hard fault and a soft fault is that soft faults can auto reset and hard faults require user intervention (usually unplugging the EV or rebooting the EVSE)