lincomatic/open_evse #152: Questions about RAPI

MrZloHex, 2022-01-27:
Hello,
I would like to firstly apologise that this is not an issue, but just a few questions about RAPI implementation, cause I haven't found any contact details to get in touch with you.
I was researching about your bit-oriented protocol RAPI and tried to implement recipient for it.
So, I got some questions:
I didn't found how to send message to start charging and should I answer on messages like _boot notification_ or _evse state transition_. 
If you can explain these things I would be very grateful!

With respect, Alexey.

lincomatic, 2022-02-06:
Have a look at the RapiSender class in this project: https://github.com/OpenEVSE/ESP32_WiFi_V4.x

closed 2022-02-06 by lincomatic

jeremypoulter, 2022-02-06:
RapiSender is part of [OpenEVSE](https://github.com/jeremypoulter/OpenEVSE_Lib) now.