lincomatic/open_evse #154: OpenEVSE start a session when there is no ESP connected

DylanVautrin, 2023-01-25:
Hello, I work on a project using OpenEVSE, my issue is linked to the powerloss behaviour, when the power come back, the Atmega boot quicker than the ESP so it start a session (about 3second) between the boot of the Atmega and the boot of the ESP, it is possible to make the Atmega "Unavailable" when there is no ESP detected ? I don't want to start a session when something is pluged and there is no ESP connected.
Thanks for helping !

KipK, 2023-01-31:
I saw the same behavior here, and would like a way to prevent Evse doing anything while Wifi module is not connected.

lincomatic, 2023-02-07:
i added $SB RAPI command. If BOOTLOCK is defined, EVSE will wait at boot until it receives $SB. You need to modify your WiFi firmware to send $SB when it boots. I did not test it, as I don't have a box to test it on at the moment. You can reopen the issue if it doesn't work properly.

closed 2023-02-07 by lincomatic

KipK, 2023-03-01:
After testing it and implemented it on Wifi fw, I think we would need a new Evse state when locked in bootlock mode. 
As it's the same state as starting now, we don't know when Evse is ready to receive the bootlock ( in the case Evs module restarts while wifi module not )

lincomatic, 2023-03-02:
EVSE always sends $AT and then $AB messages on boot. So you know when it's been rebooted. Just send $SB after receiving $AB. Also, vflags contains ECVF_BOOT_LOCKED, which can always be tested by querying vflags or from any $AT notification.

KipK, 2023-03-02:
Ho,  I've missed you've updated the vflags too. I'm going to check that thanks 👍

jeremypoulter, 2023-03-03:
@lincomatic do you plan to merge this branch to `master` ?

KipK, 2023-03-03:
@jeremypoulter , could you update the openevse lib with the new vflags?