lincomatic/open_evse #155: OpenEVSE start full amp when changing state from disable to active

KipK, 2023-01-31:
Something wrong to track down here, seems while OpenEvse si changing its state from disable to enable, it starts a charge with full amp, ignoring the Pilot value set from wifi module. Pilot is still displaying ok the requested value.

Reclaiming the charge_current will then set the charge to correct current.

I have made a little hack to mitigate this for now ( https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/532 ) but there's still a 3sec glitch where it will starts full amp. Would be better to fix this here.

lincomatic, 2023-02-07:
There is no code to change the current capacity when EVSE is enabled. You can look at the code yourself in J1772EvseController::Enable(). Maybe you're talking about during boot. I have mitigated this with BOOTLOCK

lincomatic, 2023-02-14:
OK so I will mark this closed

closed 2023-02-14 by lincomatic