lincomatic/open_evse #161: OpenEVSE don't detect plug anymore since new firmware

DylanVautrin, 2023-04-05:
Hello, i work on a project using OpenEvse, and i have an issue with the plug detection sinc i've flashed the newest firmware in the Atmega , i have two board, one in 8.2.2.T2 that don't detect when something is plugged, and another in 8.2.0.T2 that works fine. I've try approximately all the commit between 8.2.0.T2 and 8.2.2.T2 but the one's that don't detect still detect nothing when i plug something.
For now i don't want to update the other board because i'm affraid it's gonna do the same.
Do you have an idea of what going on? I can give more informations if needed.
Thanks

DylanVautrin, 2023-04-05:
I was working all the time with this repo : https://github.com/OpenEVSE/open_evse.
In the environnement emonevse (i'm in EU)

lincomatic, 2023-04-05:
Sorry I'm not the maintainer of that repo. My repo is for developers, and I don't provide support, It's best that you inquire with them. I'm not experiencing any issues w/ plug in detection w/ my own repo

DylanVautrin, 2023-04-05:
Okay thanks i'm closing this one so

closed 2023-04-05 by DylanVautrin

DylanVautrin, 2023-04-05:
Btw, i've try to use your repo but i can't find a environnement similare to the emonevse behaviour (non-tethered T2 cable)