lincomatic/open_evse #168: Stuck open contactor not annunciated properly

TechnicalLee, 2023-06-17:
When the contactor sticks open (does not close when commanded), the station reports a GROUND FAULT even though the ground is fine. The logic should be changed to display "CONTACTOR STUCK OPEN" when this is detected.

glynhudson, 2023-06-17:
That's actually not a bad idea, incorrectly reporting this as a GND fault is confusing. However, I think the wording should be something like "UNABLE TO CLOSE CONTACTOR" because the contactor not closing is more likely to be could be caused by a wiring issue rather than the contactor physically stuck.

lincomatic, 2023-06-20:
It's a limitation of the hardware. Ground Fault and a non-closing contactor generate the same signals to the firmware. This isn't fixable w/ a firmware change

chris1howell, 2023-08-04:
Correct, this is a hardware limitation. A UL change in ground monitoring from "while charging" to "continuously" will fix this limitation. The earth ground monitoring/stuck relay monitoring will be moved to separate monitoring channels. Earth ground monitoring from the input side of the relay and stuck relay on the load side. 

The load side channel will be able to determine if the relay fails to close, but earth ground is good verified by the second channel on the input side.

closed 2024-03-26 by lincomatic