lincomatic/open_evse #175: [RFC] adding a voltmeter

kolyshkin, 2024-03-27:
This is more like an RFC than a bug report, sorry for spamming issues for that.

Did some research into adding a voltmeter to openevse 5.5 hw. I am aware of OpenEVSE II but that looks like a history now.

Basically, it boils down to these options:

1. Use something like ZPMT101K (input: AC voltage, output: 0 to VCC analog signal) connected to any of ATMEGA328 analog inputs (with the most prominent candidate is PP which is connected to PC2/ADC2 and is not used in the US design), resulting something similar to OpenEVSE II design, and is already implemented (see commit  a2453c8f6bd381d48358f6055bb61e8b10c33ede). The issues are:
 - There is a need to read the ADC value a few times and figure out RMS voltage. Proper calculation is expensive, an approximate one is, well, approximate.
 - The hardware requires calibration so it's probably not very scaleable wrt mass production.
  
Those issues led me to the next option.
  
2. Use something like PZEM004t v3.0, which outputs voltage (and also current, energy, PF etc) in a digital form, so there's no need to calibrate or estimate anything. Probably more accurate, too. The issues are:
 - It needs a serial port to communicate (set to 9600 8N1), and the ATMEGA328 only have one UART which is already used for RAPI serial communication, and
 - Using software serial is costly
 - Current hardware does not have two extra pins which can be used for soft serial (I may be wrong here).
 
3. Use something like PZEM004t but connect it to ESP32 rather than atmega. ESP32 has 3 UARTs than can be mapped to any pins, and the current HW has a pinout for a second serial port (though no connector is soldered). This means, atmega's functionality of keeping track of current and energy won't be used, and ESP32 code needs to be modified to use PZEM as a source, which is an issue in itself.

4. Surely, all that can be solved by yet another microcontroller, using either PZEM or ZPMT and acting, say, as an I2C slave, but that sounds like too much hardware.

Any comments, additions, ideas etc. are very welcome. In particular, I'd like to know
- what's the main reason of not implementing a voltmeter functionality in the current hardware (I kind of tried to guess that above)
- if there are any mistakes or omissions in the above analysis.

kolyshkin, 2024-04-03:
What do you think @jeremypoulter @chris1howell @lincomatic ?

jeremypoulter, 2024-04-03:
IMHO It probably would be useful to have a volt meter built in. One interesting option not on the list would be an integration with an EmonTX4 as this would give full 3 phase power monitoring. The lack of serial on the ATMEGA would probably mean linking to the ESP32, but that being said maybe updating the ATMEGA 328 to a better MCU would be an idea, maybe merge the OpenEVSE side of things onto the EmonTX4

lincomatic, 2024-04-03:
The PZEM-004T hooked up to the ESP32 is an easy low cost solution that works well. There's no way to add a voltmeter to the ATMEGA328, because we're out of flash space. Also, since it's a safety device, it would be better not to load it down w/ more non-safety functions. Another bonus is it frees up space/CPU cycles by eliminating the ammeter code (which was always inaccurate, anyway)

kolyshkin, 2024-04-06:
TL;DR: I see two sets of contacts on the board ("Debug" and "Development"), would like to know to which ESP32 pins they are connected to, and which ones I can use for Serial1.

I started to look into how to connect PZEM004t v3.0 to OpenEVSE WiFi TFT LCD v1.0 board, and I can't find the board schematics (to find which ESP32 pins are available on which contacts).

ESP32 have three serial ports (which can be mapped to any pins).

On the software side, currently Serial and Serial2 are used for RAPI_PORT and DEBUG_PORT, and Serial1 is unused.

I can either use Serial1 or Serial2. For Serial1, I need to find two unused pins that are available from the board, and for that I need board schematics. For Serial 2, I need to completely disable debug, or change it in a way so that no serial port is used).

PS perhaps it makes sense to move this discussion to https://github.com/OpenEVSE/openevse_esp32_firmware

lincomatic, 2024-04-11:
I think there's a way to debug through a web browser, so maybe you can turn off the serial debugging. but it's up to you. I think to use Serial1, you're going to have to wire directly to the ESP32 module. Yes, this discussion should move to the ESP32 firmware side. I will close it now.

closed 2024-04-11 by lincomatic

kolyshkin, 2024-04-16:
For coherence: this continues in https://github.com/OpenEVSE/openevse_esp32_firmware/issues/825